BrillouinMicroscopy/BMicro #1: Repository BMicro; Automatisierungen einrichten

matthias-baer, 2021-01-15:


paulmueller, 2021-01-15:
FYI: To test the automation, I have to make PRs to the `main` branch, breaking our current workflow.

paulmueller, 2021-01-15:
I would like to apologize for starting with a comparatively high version number "0.1.0". It was a typo, I wanted to write "0.0.1". But since it is now already on PyPI, there is no way back.

matthias-baer, 2021-01-15:
> Antworten an: BrillouinMicroscopy/BMicro <reply@reply.github.com>

Ok, then we will rebase after you are done

paulmueller, 2021-01-15:
readthedocs, pypi, and GitHub actions are working and merged back to 1-smoke-test. The only thing missing is automatic building of installers for macOS and Windows. I will proceed with those as soon as #3 is done.

paulmueller, 2021-02-02:
Everything is in place now.

closed 2021-02-04 by matthias-baer