BrillouinMicroscopy/BMicro #2: FE: Views generieren (UIs)

matthias-baer, 2021-01-15:


raimund-schluessler, 2021-01-22:
@matthias-baer @paulmueller I moved the different views into separate ui files (only data and extraction for now) in 222c7ce6ddb4ac9f513c949853182615388a8a06. While it works, I wonder if I loaded the ui files correctly, see https://github.com/BrillouinMicroscopy/BMicro/commit/222c7ce6ddb4ac9f513c949853182615388a8a06#diff-061bb4207d2fe5862f4d1cccbdfbbdc9d9d5f0ed15a8e7257ae7750791bafe12R25-R33.

paulmueller, 2021-01-22:
This part looks good to me:
```python
ui_file = pkg_resources.resource_filename('bmicro.gui.extraction', 'extraction_view.ui')
uic.loadUi(ui_file, self) 
```
However, the initialization part could be more pythonic:

```python
def __init__(self, *args, **kwargs):
    super(ExtractionView, self).__init__(*args, **kwargs)
```

raimund-schluessler, 2021-01-22:
> However, the initialization part could be more pythonic:
> 
> ```python
> def __init__(self, *args, **kwargs):
>     super(ExtractionView, self).__init__(*args, **kwargs)
> ```

So, replace
```python
QtWidgets.QWidget.__init__(self)
```
 with
```python
super(ExtractionView, self).__init__(*args, **kwargs)
```
?

closed 2021-01-26 by matthias-baer