BrillouinMicroscopy/BMicro #4: bmlab: Service für statische Message

matthias-baer, 2021-01-15:


paulmueller, 2021-01-21:
Related to this one: I guess it would be important to have to "Checks" GitHub Actions. One that uses the latest release of bmlab from PyPI and one that uses the current "main" branch from GitHub. Any opinions?

raimund-schluessler, 2021-01-21:
> 
> 
> Related to this one: I guess it would be important to have to "Checks" GitHub Actions. One that uses the latest release of bmlab from PyPI and one that uses the current "main" branch from GitHub. Any opinions?

Wouldn't the latest release have been checked by the main branch checks action as well at some point?

paulmueller, 2021-01-21:
The thing is that the current GH Action for BMicro installs bmlab from PyPI (https://github.com/BrillouinMicroscopy/BMicro/blob/main/.github/workflows/check.yml#L30). That means it doesn't use the latest code from the bmlab main branch (only from tag releases).

matthias-baer, 2021-01-21:
Yes, that's exactly the point I wrote yesterday on Rocket Chat (and deleted later to think more about it). I guess we should have the regular checks action in bmlab, but a different kind of checks action for BMicro. In BMicro, when pushing to a branch, checks should be made based on the latest github version of bmlab on the same branch. Only when making a BMicro release, it should include the latest pypi version of bmlab instead. This would of course also mean that releases of bmlab would always have to be available before releases of BMicro could be made. Opinions?

paulmueller, 2021-01-21:
Makes sense. Should we still have a "Checks" action that runs on the latest bmlab PyPI release and is allowed to fail? Just so we know when we should increment the install_requires for bmlab in setup.py?

matthias-baer, 2021-01-21:
That makes sense. This check should also be run (and not allowed to fail) when doing a BMicro deployment action.

paulmueller, 2021-01-21:
> That makes sense. This check should also be run (and not allowed to fail) when doing a BMicro deployment action.

I just added a new action workflow that uses the bmlab repository (with the same branch name as the BMicro job runs on). However, nowI don't know how to implement what you and I are suggesting. TravisCI used to have the "allow-failure" option, which GH Actions does not apparently.

matthias-baer, 2021-01-21:
Hm... but why should we allow it to fail?

paulmueller, 2021-01-21:
Because new functionalities in 1-smoke-test will not be in the bmlab PyPI release (and then the test will fail). However, we could only run this test on the main branch.

matthias-baer, 2021-01-21:
Well, I thought when pushing BMicro to a WIP branch, we run against bmlab from github on the corresponding branch. When merging to main, we run BMicro checks against bmlab from its main branch. And when releasing BMicro to PyPi we run against bmlab from PyPi. 

Or what is your proposal?

paulmueller, 2021-01-21:
I propose:
- when pushing BMicro to a WIP branch, we run against bmlab from github on the corresponding branch
- When merging to main, we run BMicro checks against bmlab from PyPI

This would force us to have the following pipeline:
1. merge WIP for bmlab
2. release bmlab to PyPI
3. merge WIP for BMicro
4. release BMicro to PyPI etc.

This is OK I think.

If we did this: "And when releasing BMicro to PyPi we run against bmlab from PyPi.", then there would be only one check during the release (and not before the release). It makes more sense to check against PyPI bmlab on BMicro main, because we are not working on BMicro main anyway.

matthias-baer, 2021-01-21:
That's true.

closed 2021-01-26 by matthias-baer