BrillouinMicroscopy/BMicro #11: Userstory: Data View

matthias-baer, 2021-01-21:
As a user of BMicro, I want to load measurement data from a HDF5 file so that I can evaluate that data later.

Details:
The user shall be able to select the hdf5 file from an "open file dialog" triggered by a file menu. When the (single) file is selected, the application shall show information of the file as displayed in the "file" group of the screenshot below. The combobox "repetition" shall allow to select any of the available measurement series in the file. The data view is read-only. The comment shown in the screen shot is not editable.

matthias-baer, 2021-01-26:
I presume that resolution x, y, z is the attribute with the same name in the HDF file for the group 'Brillouin/0/payload', right?

matthias-baer, 2021-01-26:
@raimund-schluessler I presume that Calibration = true means that there is a calibration group under Brillouin/0/ for repetition 0?

matthias-baer, 2021-01-26:
@raimund-schluessler Where in the HDF file is "Comment" stored?

raimund-schluessler, 2021-01-26:
> 
> 
> @raimund-schluessler I presume that resolution x, y, z is the attribute with the same name in the HDF file for the group 'Brillouin/0/payload', right?

Yes, correct.

raimund-schluessler, 2021-01-26:
> 
> 
> @raimund-schluessler I presume that Calibration = true means that there is a calibration group under Brillouin/0/ for repetition 0?

Unfortunately not. This group is always present. Only if datasets are available under calibration/data then it is true.

raimund-schluessler, 2021-01-26:
> 
> 
> @raimund-schluessler Where in the HDF file is "Comment" stored?

It's an attribute on the root level of the file, named `comment`. But I think we can omit it, since it is not used at the moment and always reads "Brillouin data".

closed 2021-02-11 by matthias-baer