BrillouinMicroscopy/BMicro #17: Prepare widget for the Evaluation tab

matthias-baer, 2021-01-21:
Create a ui-file with Qt Designer / Creator for the widget containing the content of the "Peak Selection" tab. As a guideline, see the screen shot below. User whatever WIP branch is currently active.

Shadahattum, 2021-01-30:
Hey Guys, 
I have two questions: 
1)For the Validity Threshold and the refinement factor, are these always integers? do they have a max and min? I am wondering if to choose a text box for them or a spin box. 
2)What are the choices in the two drop down menus BrillouinShift_frequency and Single_peak_fit ? (sorry I don't have Matlab here)

matthias-baer, 2021-02-02:
> Hey Guys,
> I have two questions:
> 1)For the Validity Threshold and the refinement factor, are these always integers? do they have a max and min? I am wondering if to choose a text box for them or a [...]

@raimund-schluessler I guess you are the best to answer that questions....

I see that I should install MATLAB next time when I'm at the institute.... 18GB over VPN takes ages...

raimund-schluessler, 2021-02-02:
For this view, the "Open figure" and the "Masking" button, and the "Refinement factor" setting will not be necessary anymore, as they are not really used at the moment. For the masking feature, @paulmueller develops a better tool.

> 1)For the Validity Threshold and the refinement factor, are these always integers? do they have a max and min? I am wondering if to choose a text box for them or a spin box.

They are double/integers. I would use a QDoubleSpinBox. Limit for the validity threshold could be 0 to 1000.

> 2)What are the choices in the two drop down menus BrillouinShift_frequency and Single_peak_fit ? (sorry I don't have Matlab here)

The choices can be found here: https://github.com/BrillouinMicroscopy/BrillouinEvaluation/blob/master/%2BBE_Model/Model.m#L349-L362. But I would streamline the strings a bit (we can adjust this later). Maybe add just a few choices for now, e.g. Brillouin shift [GHz] and Brillouin shift [pix] or so.

raimund-schluessler, 2021-04-20:
Basic widget is implemented in 939c75608ca224936f7120ffa47757256d419db7. Stuff like double peak fit will be added later.

closed 2021-04-20 by raimund-schluessler