BrillouinMicroscopy/BMicro #19: Update the data tab ui-file

matthias-baer, 2021-01-22:


matthias-baer, 2021-01-27:
Maybe someone can have a look at the UI file. I'm not used to the designer and I'm not sure how to apply autosize constraints.

raimund-schluessler, 2021-01-27:
> 
> 
> Maybe someone can have a look at the UI file. I'm not used to the designer and I'm not sure how to apply autosize constraints.

I will have a look.

matthias-baer, 2021-02-02:
> > Maybe someone can have a look at the UI file. I'm not used to the designer and I'm not sure how to apply autosize constraints.
> 
> I will have a look.

Thanks, but never mind. I found out how to do it. :-)

raimund-schluessler, 2021-02-02:
> 
> 
> > > Maybe someone can have a look at the UI file. I'm not used to the designer and I'm not sure how to apply autosize constraints.
> > 
> > 
> > I will have a look.
> 
> Thanks, but never mind. I found out how to do it. :-)

Ok, great. Sorry I hadn't had the time to do it.

raimund-schluessler, 2021-02-11:
@matthias-baer Should we close this issue now that it's in Done?

matthias-baer, 2021-02-11:
I think so.

closed 2021-02-11 by matthias-baer