BrillouinMicroscopy/BMicro #23: Backend: perform rotations and mirroring

matthias-baer, 2021-01-22:


matthias-baer, 2021-01-27:
@raimund-schluessler The images in the Water.h5 file have the shape (2, 400, 400). What is the first axis?

raimund-schluessler, 2021-01-27:
> 
> 
> @raimund-schluessler The images in the Water.h5 file have the shape (2, 400, 400). What is the first axis?

That is the number of images we acquire per measurement point. It's quite convenient to store them like this, because HDF Viewer lets you easily navigate through them.

The same is true for the calibration images, there the first axis also is the number of images (usually 10).

Tianbe, 2021-01-31:
So I guess we should have functions that perform the rotation/mirroring of the images. I guess this should then just work for one single image? I would then just check if the input comes in a n x m numpy array and return the tilted image.

matthias-baer, 2021-02-01:
Or alternatively you could apply the rotation/reflection to all images, if the numpy array has three dimensions and only to one if it has two dimensions...

Tianbe, 2021-02-02:
I think that comes down to the question how we would like to set up the the analysis pipeline. I guess in the end we anyway have to loop over all images before they are analyzed an then we would just apply the rotation before. We often also have more than three dimensions in our data array: three spacial ones, another dimension for multiple images at the same position and eventually also another one for fitting multiple peaks. So I would keep these functions simple to just rotate a 2-d image, but if there are objections, I could of course also immediately apply it to all images. What do you think?

paulmueller, 2021-02-02:
I would keep it as simple as possible, i.e. choose the solution which is least prone to human error. In terms of speed, I would not worry much - e.g. numpy.rot90 just returns a view of the data (https://scipy-cookbook.readthedocs.io/items/ViewsVsCopies.html), so there is nothing copied or computed in memory. It would probably best to have one function that does all the transformations (rotate, flip, etc).

I would implement it like so (assuming data is stored as HDF5):

```python

def load_data(path, rotate=0, flip_x=False, flip_y=False, ...):
    with h5py.File(path, "r") as h5:
        data = h5["some_data"][:]
    if rotate:
        data = np.rot90(m=data, k=rotate)
    if flip_x:
        data = ....
    return data
```

matthias-baer, 2021-02-02:
I think it would be good to keep file loading and image operations separated.  So I think we should have something with the signature

```
def set_orientation(img_data, rotate=0, flip_x, flip_y)
```

And the img_data argument is simply a numpy array obtained from „elsewhere“. Regarding file loading, since the hdf data files seem to have a complex internal structure, I would make that abstract. That‘s why the data loading so far has been implemented by dedicated classes (which of course could still be improved to be more user friendly).

Tianbe, 2021-02-03:
That sounds reasonable and is also what I had in mind. I already started to implement something with np.rot90 und np.flipud

Tianbe, 2021-02-03:
...but then I would still assume that img_data is just one image, because if img_data contains up to five dimensions (but not always), I don't know how to perform the rotations. Accordingly, I would handle all the loading and also identification how img_data looks like somewhere else.

raimund-schluessler, 2021-02-03:
> ...but then I would still assume that img_data is just one image, because if img_data contains up to five dimensions (but not always), I don't know how to perform the rotations. Accordingly, I would handle all the loading and also identification how img_data looks like somewhere else.

img_data has maximum three dimensions. Two for the x- and y-axis respectively, the third for the number of images at one position.

What has more dimensions (and this always) is the results array. This has three spatial dimensions, another one for the number of images per point, one for the number of peaks per fit and one for the Stokes- and Anti-Stokes peaks (so 6 in total). But is is unrelated to the images and their rotation/flipping.

raimund-schluessler, 2021-02-03:
In the Matlab code, I wrote a function which automatically rotates/flip/mirrors all images correctly, without the need for specifying how to correctly do it, see https://github.com/BrillouinMicroscopy/BrillouinEvaluation/blob/master/%2BBE_Controller/Data.m#L782-L814

I think that is a good idea, because all images need to be handled the same way. Manually having to supply the correct parameters seems cumbersome, as they are always the same for one evaluation file. Hence, I would prefer a function that wraps these parameters and just gets them from the settings.

Tianbe, 2021-02-03:
> Manually having to supply the correct parameters seems cumbersome, as they are always the same for one evaluation file

But you still have to select how you rotate the image in the Gui, right? And then you perform that operation on every image. So I would have one function applying the operation (here its called adjustOrientation) on a single image and another function supplying the correct parameters and applying adjustOrientation on each image. So thats how I would do it then

raimund-schluessler, 2021-02-03:
> But you still have to select how you rotate the image in the Gui, right?

Yes, there needs to be an option in the GUI to adjust the settings.

> And then you perform that operation on every image. So I would have one function applying the operation (here its called adjustOrientation) on a single image and another function supplying the correct parameters and applying adjustOrientation on each image. So thats how I would do it then

Well, I don't know exactly how that should be solved in python. In C++ you would have a class `image` with a public function, let's call it `getRotatedImage(<specify which image you want, e.g., by index>)` that returns an image correctly orientated. This function would be called throughout the program and you wouldn't need to care about the implementation of the function.

You could of course have a private function which is called by `getRotatedImage` which rotates the image and needs parameters how to do that. But this should only be called from within the `image` class and never be exposed to the rest of the program.

matthias-baer, 2021-02-03:
Timon, don't worry too much about it. Choose whatever makes most sense to you right now. If it turns out to be impracticable later, it can still be refactored.

Tianbe, 2021-02-03:
Sounds good. I will push something today.

Tianbe, 2021-02-03:
I pushed now my branch (I created a different branch for my changes). I created a pull request, but realized, that 7 checks where cancelled and 5 failed, which is on average not so good I guess -.-

closed 2021-02-04 by matthias-baer