BrillouinMicroscopy/BMicro #33: (Story #27): Fit and show circle through selected points

matthias-baer, 2021-02-04:


Tianbe, 2021-02-09:
Fitting and showing sounds like one could split that up in two tasks, one for bmlab (fitting), and one for bmicro (showing the circle)? Or are this too small tasks? On could then also think of putting together the task for fitting and optimizing the fit, which is a quite similar task?

matthias-baer, 2021-02-10:
My feeling was that splitting it up would be too small, but one never knowns in advance. In principle one could split it up. Or leave it as a single issue with more than people working on it.

Tianbe, 2021-02-10:
I am still not really familiar with the gui programming, if someone could take care of this part. I could try the fitting stuff beginning of next week

matthias-baer, 2021-02-11:
No problem, you can concentrate on the fitting part. The GUI part for selecting the peak points is done so far.

Tianbe, 2021-02-12:
OK, cool then I will start to work on that

matthias-baer, 2021-02-16:
@Tianbe I have inserted a dummy implementation in `bmlab.image` so that you can see what the front end expects. The `points` argument is a list of (x,y) tuples.

Tianbe, 2021-02-18:
Ok, thanks. I did not really find time this week, to work on this. But I will do that now

matthias-baer, 2021-02-22:
I see you have pushed an implementation and tests. Thanks!

Just a quick comment: when you use random numbers in the test suite, it is advisable to use a specific random number seed so that the tests always run with the same set of random numbers.

Tianbe, 2021-02-22:
Sounds like a good idea. I also still have hard coded start values for the optimization, which might me not optimal

Tianbe, 2021-02-22:
> 
> 
> I see you have pushed an implementation and tests. Thanks!
> 
> Just a quick comment: when you use random numbers in the test suite, it is advisable to use a specific random number seed so that the tests always run with the same set of random numbers.

Do you want to adapt this, with the seed, or shall I do that?

matthias-baer, 2021-02-22:
If you have time to do it, that would be great. And an "analytic" test case would also be nice in addition, i.e. given a circle (center+radius), select three points (without noise), then fit, and the result should be almost the circle you started with.

Tianbe, 2021-02-22:
Ok, I will refine it according to your suggestions

matthias-baer, 2021-02-23:
I see that the current implementation of the fit only fits for the center coordinates. But it should also fit for the radius of the circle. That should lead to a more robust fit than the averaging approach...

raimund-schluessler, 2021-02-23:
> 
> 
> I see that the current implementation of the fit only fits for the center coordinates. But it should also fit for the radius of the circle. That should lead to a more robust fit than the averaging approach...

Yes, it has to fit the radius as well.

Tianbe, 2021-02-23:
Ok, I will check how it was done in the Matlab version and adapt it

Tianbe, 2021-02-23:
According to the previous software I now parametrized the y_coordinates of a circle with the following function: ```
def circle_model(r, x_c, y_c, x):
    with np.errstate(invalid='ignore'):
        return y_c - np.sqrt(r ** 2 - (x - x_c) ** 2)
``` The issue I have now that I don't know ho to deal with the negative values in the square root and also how to deal with the two possible solutions of the quadratic function. I think Raimund just always used the "minus-solution", but i was not sure where that comes from. Any suggestions?

Tianbe, 2021-02-23:
According to the previous software I now parametrized the y_coordinates of a circle with the following function:
 ```
def circle_model(r, x_c, y_c, x):
with np.errstate(invalid='ignore'):
return y_c - np.sqrt(r ** 2 - (x - x_c) ** 2) 
``` 
The Issue that I have with that, is that I don't know how to deal with the negative values in the square root (gives me nan) and also hoow to deal with the two different solutions of the quadratic function. Raimund always just used one (if I got this correctly), but I don't really know where this comes from. Any suggestions?

matthias-baer, 2021-02-23:
As a "cost" function to optimize, I would use the square instead, i.e. minimize

sum( (x-center_x)**2 + (y-center_y)**2 - r**2) 

with respect to center_x, center_y, r.

Tianbe, 2021-02-23:
Ok, sounds good. I used ```
def circle_opt(c, x_coord, y_coord):

    return np.nansum((circle_model(*c, x_coord) - y_coord)**2)
```
but that did not really work out

matthias-baer, 2021-02-23:
Note my edit. I forgot another `**2`.

Tianbe, 2021-02-23:
> 
> 
> As a "cost" function to optimize, I would use the square instead, i.e. minimize
> 
> ```
> sum(( (x-center_x)**2 + (y-center_y)**2 - r**2) **2 )
> ```
> 
> with respect to center_x, center_y, r. That avoids issues with imaginary square roots and multiple solutions.

That's the better solution in any means. I will adapt everything and push that today.

matthias-baer, 2021-02-23:
Have you pulled the latest version of bmlab? It uses skimage now at some place.

Tianbe, 2021-02-23:
> 
> 
> I get `ModuleNotFoundError: No module named 'skimage'` when trying to run the program now.

You have to install it. Actually pycharm should offer you to install it

Tianbe, 2021-02-23:
The other option is to use "pip install skimage"

raimund-schluessler, 2021-02-23:
PyCharm offered to install dependencies and I did, but still the error message is there.

It is the latest state of both projects on branch 27-basic-...

matthias-baer, 2021-02-23:
Did you submit "pip install -e . " in the bmlab directory?

raimund-schluessler, 2021-02-23:
The reason was that I have two `venv`s one for BMicro and one for bmlab. And the required packages are only installed in the corresponding `venv`. So when running BMicro, required packages for bmlab where missing in the BMlab `venv`. 🙈

Tianbe, 2021-02-23:
To sum it up, it works with Matthias approach, but for some reason the performance for few data points without noise could be better somehow (but thats only a test case, not sure if this is so important now). Maybe I push it on a separate branch and we can have a look again?

matthias-baer, 2021-02-24:
Well, if it works then you can also push it on the current development branch unless the performance problems are severe.

matthias-baer, 2021-02-25:
@Tianbe Thanks for the latest push! I had to modify it slightly because it broke one of my new tests.

Tianbe, 2021-02-25:
Ok, I was already wondering why that happened 😁 God example how you can break code that you did not even work on

closed 2021-02-26 by raimund-schluessler