BrillouinMicroscopy/BMicro #34: Allow to save current evaluation progress

raimund-schluessler, 2021-02-12:
We should be able to save the current evaluation progress. So everything a user has selected and evaluated so far.

- Data: the rotation, reflection and selected setup
- Extraction: Selected points for each calibration
- Calibration: Selected peaks and calculated calibrations
- Peak selection: peaks selected
- Evaluation: All parameters extracted

@matthias-baer Please edit the issue as necessary. Which user story should we assign this to? I think it would be helpful for developing/debugging already, since you don't have to manually put all settings again, but can just load the evaluated data.

matthias-baer, 2021-02-12:
Since it is useful for all later tasks, I would add it to the current user story implement only the saving of the evaluation process up to this point. For the next steps, we then update the saving procedure for the newly added functionality.

matthias-baer, 2021-02-12:
@raimund-schluessler Can you add a description where the progress file shall be saved relative to the h5 file? You showed me the folder structure, but I can't remember in detail.

closed 2021-02-16 by matthias-baer

raimund-schluessler, 2021-02-16:
@matthias-baer What fileformat is `bms`? Can we save the data as `h5` maybe? Otherwise it gets complicated to open this with other programs (including Matlab).

matthias-baer, 2021-02-16:
It was the easiest solution to serialize the session with `pickle`. But  it can be changed to h5 of course.

matthias-baer, 2021-02-16:
I opened a new issue #41 for changing the file format.