BrillouinMicroscopy/BMicro #36: (User Story #35) Show extracted spectrum for selected calibration dataset

matthias-baer, 2021-02-12:


matthias-baer, 2021-03-01:
@raimund-schluessler I changed the implementation for the extraction from non-interpolating rectangles to interpolating lines:

[image]

The extracted spectrum for one frame of one calibration image (no averaging over frame is ok?) looks like this:

[image]

Does that look plausible to you?

raimund-schluessler, 2021-03-01:
@matthias-baer Thanks a lot, this looks good. Averaging over multiple images is not necessary. We will fit the spectrum separately for each image.

closed 2021-04-13 by matthias-baer