BrillouinMicroscopy/BMicro #37: (User Story #35) Select spectral ranges for Brillouin peaks

matthias-baer, 2021-02-12:


matthias-baer, 2021-03-01:
When selecting spectral ranges in the Matlab version, do you select each Brillouin peak separately or do you select one Brillouin region as double peak?

raimund-schluessler, 2021-03-01:
> @raimund-schluessler When selecting spectral ranges in the Matlab version, do you select each Brillouin peak separately or do you select one Brillouin region as double peak?

We select the whole region including both Brillouin peaks (for water and methanol). You also have to fit a double-Lorentz peak instead of two single Lorentz peaks. Since the peaks overlap, fitting two separate peaks would bias the fit towards the other peak, respectively.

matthias-baer, 2021-03-01:
That's why I'm asking. Fitting each peak individually would be quite unstable.

closed 2021-03-02 by matthias-baer