BrillouinMicroscopy/BMicro #41: Change session file format from pickle to HDF

matthias-baer, 2021-02-16:


raimund-schluessler, 2021-04-30:
@matthias-baer Loading a session does not seem to restore the extraction and calibration models (or the view is not updated, need to check).

And saving a loaded session crashes (Open program, load file without session file, do some evaluation, save session, close program, load session, save again -> crash).

raimund-schluessler, 2021-04-30:
See https://github.com/BrillouinMicroscopy/bmlab/pull/10 for a test showing the issue.

matthias-baer, 2021-05-04:
Hm... I have pushed a fix to the branch fix-issue-41. Locally all tests run fine, but for some strange reason not on github actions. Have to check.

raimund-schluessler, 2021-05-04:
> 
> 
> Hm... I have pushed a fix to the branch fix-issue-41. Locally all tests run fine, but for some strange reason not on github actions. Have to check.

Maybe loading the file https://github.com/BrillouinMicroscopy/bmlab/blob/fix-issue-41/tests/test_serializer.py#L62 doesn't work with the path being different locally and on Github maybe?

matthias-baer, 2021-05-05:
No it was not due to the path. There were several issues. Somehow the Windows version of h5py seems not to reliably releases the handle when leaving the context. At least the closing and the tidy up process were conflicting. But only for Windows. 

Then there was another issue for Ubuntu only, where h5py seems to read strings as binary strings instead of utf-8. Not so on Mac and Windows.

matthias-baer, 2021-05-05:
Anyways, should be solved now.

closed 2021-05-05 by matthias-baer