BrillouinMicroscopy/BMicro #43: Closing a file should reset all tabs

raimund-schluessler, 2021-02-17:
The data in the calibration tab is currently not reset when closing a file. It is still possible to select different calibration from the now closed file, leading to a program crash.

matthias-baer, 2021-02-17:
Fixed clearing the extraction tab. But I not see that the orientation controls on the data tab should probably (?) also be reset to the default values.

matthias-baer, 2021-02-17:
Closed with commit 5dd08f2

closed 2021-02-17 by matthias-baer

raimund-schluessler, 2021-02-17:
> But I not see that the orientation controls on the data tab should probably (?) also be reset to the default values.

I don't have a strong opinion here. I am fine with the current behaviour, as long as the user cannot make the app exit unexpectedly.

commit 5dd08f2d (2021-02-17): Fix #43