BrillouinMicroscopy/BMicro #44: Allow to adjust extraction granularity

matthias-baer, 2021-02-25:


raimund-schluessler, 2021-02-25:
Not sure this is necessary. In the Matlab version, it is possible to adjust these values, but we basically never use that option. So I think default values would be fine (at least for now).

But what needs to be checked, is that the same values as in the Matlab version are used.

matthias-baer, 2021-02-25:
Yes, that's probably not the case yet. Also, I may have implemented a different approach for extraction, so let's discuss that tomorrow.

Tianbe, 2021-04-01:
Did we now decide to implement this? If yes I could work on that

Tianbe, 2021-04-01:
Is the given example code actually still there? Could not find this piece

raimund-schluessler, 2021-04-01:
> 
> 
> Is the given example code actually still there? Could not find this piece

Probably not. Give me some minutes, I currently work on the code you would need to adjust. However, I am not sure this function is critical at the moment.

raimund-schluessler, 2021-04-01:
The width of the extraction arc is set here https://github.com/BrillouinMicroscopy/bmlab/blob/35-basic-calibration/bmlab/models/extraction_model.py#L89-L90.

But before you implement a GUI for changing the value, I would propose to make BMicro actually showing the extraction positions from bmlab. At the moment, the extraction positions shown in BMicro do not respect settings in bmlab but are calculated in BMicro as seen here:

https://github.com/BrillouinMicroscopy/BMicro/blob/35-basic-calibration/bmicro/gui/extraction/extraction_view.py#L158-L168

raimund-schluessler, 2021-04-07:
> But before you implement a GUI for changing the value, I would propose to make BMicro actually showing the extraction positions from bmlab. At the moment, the extraction positions shown in BMicro do [...]

Done in 76e8f2c67009fb39662d4375e6d856b14d9bd76f

closed 2021-05-20 by raimund-schluessler