BrillouinMicroscopy/BMicro #45: Merge overlapping regions when selecting spectral ranges.

matthias-baer, 2021-02-25:


Tianbe, 2021-03-26:
I wonder if this is actually necessary and also useful. If I for example accidentally overlap the region of the Brillouin peaks with the Rayleigh peaks, why should I merge these regions? Then I would end up with a mixed Brillouin-Rayleigh region...

raimund-schluessler, 2021-03-26:
> I wonder if this is actually necessary and also useful. If I for example accidentally overlap the region of the Brillouin peaks with the Rayleigh peaks, why should I merge these regions? Then I would end up with a mixed Brillouin-Rayleigh region...

Merging overlapping Brillouin and Rayleigh regions doesn't make sense. But merging overlapping Brillouin (or Rayleigh) regions is fine.

Tianbe, 2021-03-26:
So in case I would like would like to extend the selected region. Then I would not have to clear the selection again. So I guess that boils down to adjusting CalibrationModel.add_brillouin_region() and check for overlaps, before the new region is appended to the brillouin_regions. I guess i could take care of that

raimund-schluessler, 2021-04-01:
Implemented with https://github.com/BrillouinMicroscopy/bmlab/commit/83b35bb6638ad6750c10cdfd2a285cf174d8fb32

closed 2021-04-01 by raimund-schluessler