BrillouinMicroscopy/BMicro #46: BMicro is blocking builds for my other projects on Appveyor

paulmueller, 2021-02-26:
Hey there,

the BMicro Appveyor build is connected to my account. The problem is that there is only one concurrent build per account and this blocks my other builds, especially because now the build won't finish because of a pyinstaller-issue (which I have no time to debug right now).

Would someone of you @raimund-schluessler ? @matthias-baer ? be willing to connect your GitHub account to Appveyor? There is nothing much to do. I believe once I cut my connection to the BrillouinMicroscopy organization and you add it. I should be fine...

Thanks!

raimund-schluessler, 2021-02-26:
I am fine with doing so, but could you give me a hint on how to?

paulmueller, 2021-02-26:
Basic setup:

1. Go to https://ci.appveyor.com and login with your GitHub account (you have to give them access to your public repositories)
2. Allow Appveyor access to the BrillouinMicroscopy organization and add a new project BMicro

Setting up GitHub releases:

1. Go to https://github.com/settings/tokens and create a new token. Name it "deployment to releases with appveyor". Only check "public_repo" permissions.
2. Copy the token, go to https://ci.appveyor.com/tools/encrypt and encrypt it. Replace my token with yours in the .appveyor file: https://github.com/BrillouinMicroscopy/BMicro/blob/main/.appveyor.yml#L76

raimund-schluessler, 2021-02-26:
Done. I pushed it to the 27-basic-extraction branch. Should I also commit to the main branch?

paulmueller, 2021-02-26:
Thanks! No need for editing master. Could you please post the project link? I will update the badge.

raimund-schluessler, 2021-02-26:
I guess it should be https://ci.appveyor.com/project/raimund-schluessler/BMicro

commit 38e9d35a (2021-02-26): update appveyor badges (close #46)
closed 2021-02-26 by paulmueller