BrillouinMicroscopy/BMicro #48: get_circle_fit needs to accept time instead of cal_key

raimund-schluessler, 2021-03-08:
Currently `get_circle_fit` accepts `cal_key` as parameter and returns the exact CircleFit for the requested calibration. However, for the evaluation, we need a function that returns the CircleFit based on the time when the image was acquired. For times inbetween calibrations, the CircleFit needs to be interpolated. This function should also be used for the Calibration view.

matthias-baer, 2021-03-08:
That's true. However, it is not a bug, but a new feature, because that was not needed until recently.

I'm sorry, I haven't had time for the project last week and probably also not much this week. Please don't wait for me to carry on with the work.

raimund-schluessler, 2021-03-08:
> That's true. However, it is not a bug, but a new feature, because that was not needed until recently.

Fine with me 🙂 But I would also use it for the calibration, so `get_circle_fit(cal_key)` won't be needed anymore, I think.

> I'm sorry, I haven't had time for the project last week and probably also not much this week. Please don't wait for me to carry on with the work.

Yes, I will try to work a bit on it this week.

raimund-schluessler, 2021-04-08:
Implemented with https://github.com/BrillouinMicroscopy/bmlab/commit/9c380ef0ded0f3a721cac168d1d6f244ceb1d1e2

closed 2021-04-08 by raimund-schluessler