BrillouinMicroscopy/BMicro #49: Move session into bmlab

raimund-schluessler, 2021-03-16:
I have the impression, that `session.py` https://github.com/BrillouinMicroscopy/BMicro/blob/main/bmicro/session.py should rather be located in `bmlab` than in `BMicro`. Otherwise it might be impossible to have `bmlab` evaluate a file headless on its own.

What do you think @matthias-baer?

matthias-baer, 2021-03-16:
Yes, I agree. It should be migrated to bmlab because of headless applications without the GUI. Then we should also re-consider if it should still be a singleton (probably not). Not being a singleton would allow to have parallel sessions in headless scenarios. But then we would have to be more careful in the GUI regarding memory and resource leakage e.g. when clearing a session.

raimund-schluessler, 2021-04-01:
I think being a singleton is fine. If you want to evaluate multiple files, you can evaluate them sequentially or run multiple processes in parallel. Having multiple sessions in the same process doesn't make much sense to me.

raimund-schluessler, 2021-04-07:
Implemented in 632d72e3c14da071862320a4823ca4e860b2758a

commit 632d72e3 (2021-04-01): fix: move session from BMicro to bmlab

Closes #49.
closed 2021-04-07 by raimund-schluessler