BrillouinMicroscopy/BMicro #50: Lorentz fit is unstable

matthias-baer, 2021-04-12:
The current implementation of the Lorentz fit is not stable. In particular for data from actual images, the fit fails now. The data extracted and to be fitted looks like this:

[image]

I have implemented a test that shows the failing of the fit (in `bmlab/tests/test_fits.py`)

```
def _test_fit_spectral_range():
    """ The data for this test case has been extracted manually from the running
        BMicro application.
    """

    data_dir = pathlib.Path(__file__).parent / 'data'

    region = np.load(data_dir / 'rayleigh_reg0_region.npy')
    xdata = np.load(data_dir / 'rayleigh_reg0_xdata.npy')
    ydata = np.load(data_dir / 'rayleigh_reg0_ydata.npy')

    fit_lorentz(xdata[range(*region)], ydata[range(*region)])

    # import matplotlib.pyplot as plt
    # plt.plot(xdata[range(*region)], ydata[range(*region)])
    # plt.show()
```

raimund-schluessler, 2021-04-12:
I think I fixed this on Friday, but forgot to push. 🙈 Let me merge my changes with your test and check if this fixes it.

matthias-baer, 2021-04-12:
Ah, great. Then you now have an additional test case that comes from a real microscopy file. :-)

raimund-schluessler, 2021-04-12:
I pushed my changes. Could you check again, the test doesn't seem to run automatically.

matthias-baer, 2021-04-12:
Yes, because I commented it out by the leading underscore.

raimund-schluessler, 2021-04-12:
Could it be that the data file is missing from your commit? The test seems to fail loading the data as far as I can tell.

matthias-baer, 2021-04-12:
RIght, I forgot to add it. 🙈

matthias-baer, 2021-04-12:
Ok, pushed and test is running.

closed 2021-04-12 by matthias-baer