BrillouinMicroscopy/BMicro #51: Clarification of frequency axis

matthias-baer, 2021-04-13:
@raimund-schluessler  I'm not sure about the direction of increasing frequency. Is it from left to right or from right to left in the picture below?

[image]

raimund-schluessler, 2021-04-13:
Higher frequencies are towards the right side in this image (I flipped the axis in https://github.com/BrillouinMicroscopy/bmlab/commit/5f6bc676ad6d21b020c81118f41e9ba4eee220b2 to make it like this).

matthias-baer, 2021-04-13:
I may have flipped this once again. ;-) To make sure: on the plot above assuming low pixels are low frequencies and high p high f. it the order right or wrong?

raimund-schluessler, 2021-04-13:
> 
> 
> I may have flipped this once again. ;-) To make sure: on the plot above assuming low pixels are low frequencies and high p high f. it the order right or wrong?

Yes, I saw that 😄 In BMicro it is now shown wrongly. The side where the Brillouin and Rayleigh peaks are closer together (here the right side) is the side with higher frequencies. So the left Rayleigh peak is at 0 GHz and the right one at approx. 15 GHz.

closed 2021-04-20 by raimund-schluessler