BrillouinMicroscopy/BMicro #56: Peak selection tab not updated automatically

raimund-schluessler, 2021-04-22:
The peak selection tab (tables and plot) doesn't update automatically yet. One has to click e.g. `Clear` in order to see the plot. This needs to be changed, so that the plot always shows the latest data.

The question is where we should handle this. E.g. when the extraction or calibration data changes, the peak selection tab has to update when it is viewed. Can we just trigger a refresh when one selects the peak selection tab? @matthias-baer Input welcome 🙂

matthias-baer, 2021-04-22:
Can you add a screenshot for better visualization, please?

raimund-schluessler, 2021-04-22:
> 
> 
> Can you add a screenshot for better visualization, please? (Pasting from clipboard works here)

You can even have a GIF 😉 

[image]

matthias-baer, 2021-04-22:
Cool. :-)

How about that: let's have a controller class (e.g. `WorkflowController`) for the complete workflow which operates on the model classes. WorkflowController should be the only one with direct access to the model instances (in particular write, but better also read). When some data in the model changes, it is only done through the controller and since it knows the dependent tasks and corresponding data, it can update/clear all data of downstream tasks. As a first version, I would probably just let WorkflowController delete the data of downstream tasks, so that when a tab corresponding to some downstream task is activated, it's just in the initial state. Later one can think about keep part of the downstream data (e.g. selection of regions), but that would make it more complicated for the beginning.

raimund-schluessler, 2021-04-22:
> Cool. :-)
>
> How about that: let's have a controller class (e.g. `WorkflowController`) for the complete workflow which operates on the model classes. WorkflowController should be the only one with [...]

I already created controllers for every model, which handle tasks mainly related to the corresponding model: https://github.com/BrillouinMicroscopy/bmlab/tree/35-basic-calibration/bmlab/controllers

Maybe we can combine the two approaches?

matthias-baer, 2021-04-22:
> Maybe we can combine the two approaches?

Yes, at the moment I think it would be easier to have one controller for the workflow instead of one for each task, because otherwise you would need another layer for orchestrating the different controllers. Maybe we will need that later if the class becomes too big, but there is probably no need for the overhead yet.

raimund-schluessler, 2021-04-22:
> Yes, at the moment I think it would be easier to have one controller for the workflow instead of one for each task, because otherwise you would need another layer for orchestrating the different [...]

Ok. Fine with me. Feel free to refactor this.

raimund-schluessler, 2021-05-04:
Closed with https://github.com/BrillouinMicroscopy/BMicro/pull/76.

closed 2021-05-04 by raimund-schluessler