BrillouinMicroscopy/BMicro #62: BMicro crashes when selecting different tab while evaluating

raimund-schluessler, 2021-04-29:


maroba, 2021-04-30:
I will probably have some time this afternoon for looking into the current issues...

matthias-baer, 2021-04-30:
I will probably have some time this afternoon to look into the current issues...

raimund-schluessler, 2021-04-30:
> 
> 
> I will probably have some time this afternoon to look into the current issues...

Great. I will likely merge the 35-basic-calibration branch today, since the most important features are present (although sometimes a bit buggy still). So I think it makes sense to then create a branch per issue that is being worked on.

raimund-schluessler, 2021-05-04:
The problem is that the whole UI is recreated when selecting a different tab. This also resets the repetition combobox which leads to the `_current_repetition_key` being an empty string (as long as the UI is updated). Hence https://github.com/BrillouinMicroscopy/bmlab/blob/main/bmlab/session.py#L40-L44 returns `None` and the calibration aborts.

The whole UI being reset and recreated is probably also the reason why it takes so long to switch between tabs.

PR #76 (2021-05-04): Only update necessary tab when switching tabs
This changes the UI so that only the necessary tab is updated when switching tabs. This speeds up switching between tabs, prevents the program from crashing when switching tabs while evaluating and ensure the correct data is shown for each tab.

Closes #62 and #56.

Todo:
- [x] Reset calibration, peak-selection and evaluation tab when a file is closed
closed 2021-05-04 by raimund-schluessler