BrillouinMicroscopy/BMicro #79: Peak selection plot not showing after loading session

raimund-schluessler, 2021-05-05:
The problem is, that `circle_fits_interpolation` in the extraction_model is not recreated after opening a file.

matthias-baer, 2021-05-05:
Ah, right. I will implement some kind of post-creation hook.

raimund-schluessler, 2021-05-05:
Great. Maybe it's enough to call a refresh/recreate function on every model after the data was loaded.

Other than that the serialization seems to work well. Evaluating a file and saving the data works fine.

When this post-creation hook is available we should also look into moving the calibration interpolation out of the function call and do it only once when the underlying data changes:
https://github.com/BrillouinMicroscopy/bmlab/blob/main/bmlab/models/calibration_model.py#L183-L197
https://github.com/BrillouinMicroscopy/bmlab/blob/main/bmlab/models/calibration_model.py#L151-L161

matthias-baer, 2021-05-05:
The Serializer class now has a method `post_deserialization` which is empty in the base class and is called after deserialization. When child classes need to do stuff after deserialization, they can override it. I did that in the ExtractionModel to refresh the interpolations. 

It should work, but maybe there is some even nicer way to do that with decorators like in Java EE.

matthias-baer, 2021-05-05:
Actual issue is fixed with PR #21.

Open: moving interpolation stuff to controller.

raimund-schluessler, 2021-05-05:
> 
> 
> Actual issue is fixed with PR #21.
> 
> Open: moving interpolation stuff to controller.

Great, thanks a lot.

I created a separate issue for the controller stuff https://github.com/BrillouinMicroscopy/bmlab/issues/22

closed 2021-05-05 by raimund-schluessler