BrillouinMicroscopy/BMicro #93: Updating plot during evaluation is slow

raimund-schluessler, 2021-05-20:
I assume something is off with the threading. Because updating the plot while the evaluation is not running is very fast.

paulmueller, 2021-05-24:
Note that matplotlib is not thread-safe (which might be an issue here somehow).

A first analysis shows that the call `self.mplcanvas.draw()` is responsible (takes more time during evaluation).

paulmueller, 2021-05-24:
Could it be that updating the plot is slower, *because* the evaluation is running? On my machine I only see 1 CPU at 100%.

paulmueller, 2021-05-24:
I will attempt to explain what is going on. BMicro runs in one process. If you use threading, then there are multiple threads in that process. The process is executed by one CPU core. If you have heavy computation going on in one thread, then the other thread will automatically become slower as well, because there is only one CPU core doing all the work.

More backgound information (also about GIL) is e.g. here: https://www.guru99.com/python-multithreading-gil-example.html

raimund-schluessler, 2021-05-25:
Different threads can in principle run on different cores. Maybe there is a problem with the processor affinity of the threads, which causes them to run on the same core. Qt should actually take care of this automatically. I will try to have a look.

raimund-schluessler, 2021-07-05:
I have the suspicion that the global interpreter lock of Python effectively prevents running multiple threads with Python code simultaneously. It seems that only the thread which currently holds the GIL is allowed to run at a time.

So moving to multiprocessing might really be the only option.