BrillouinMicroscopy/BMicro #112: cannot open bmicro v0.1.3

juko1, 2022-02-15:
Last login: Tue Feb 15 13:32:29 on ttys001

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
01-guc-grp-0050:~ jkolb$ python3 bmicro
/Library/Frameworks/Python.framework/Versions/3.9/bin/python3: can't open file '/Users/jkolb/bmicro': [Errno 2] No such file or directory
01-guc-grp-0050:~ jkolb$ python3 -m bmicro
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bmicro/__main__.py", line 8, in <module>
    from bmicro.gui.main import BMicro
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bmicro/gui/main.py", line 11, in <module>
    from . import calibration
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bmicro/gui/calibration/__init__.py", line 2, in <module>
    from .calibration_view import CalibrationView
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bmicro/gui/calibration/calibration_view.py", line 13, in <module>
    from bmlab.controllers.calibration_controller import CalibrationController
ModuleNotFoundError: No module named 'bmlab.controllers.calibration_controller'; 'bmlab.controllers' is not a package
01-guc-grp-0050:~ jkolb$

raimund-schluessler, 2022-02-15:
@juko How did you install BMicro?

juko1, 2022-02-15:
Created a virtual environment and installed it there

raimund-schluessler, 2022-02-16:
@juko1 Please try using the MacOSX dmg file from the latest release:
https://github.com/BrillouinMicroscopy/BMicro/releases/tag/0.1.5

juko1, 2022-02-17:
01-guc-grp-0050:Desktop jkolb$ python3 -m pip install BMicro
Requirement already satisfied: BMicro in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (0.1.3)
Requirement already satisfied: h5py>=2.10.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (3.1.0)
Requirement already satisfied: scipy>=0.14.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (1.6.0)
Requirement already satisfied: bmlab in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (0.0.11)
Requirement already satisfied: matplotlib in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (3.5.1)
Requirement already satisfied: numpy>=1.17.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (1.19.5)
Requirement already satisfied: pyqt5 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (5.15.2)
Requirement already satisfied: pyqtgraph in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (0.12.1)
Requirement already satisfied: scikit-image in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from BMicro) (0.19.1)
Requirement already satisfied: packaging>=20.8 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from bmlab->BMicro) (21.0)
Requirement already satisfied: shapely>=1.8.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from bmlab->BMicro) (1.8.0)
Requirement already satisfied: pytest-mock in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from bmlab->BMicro) (3.7.0)
Requirement already satisfied: pytest in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from bmlab->BMicro) (7.0.1)
Requirement already satisfied: cycler>=0.10 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from matplotlib->BMicro) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from matplotlib->BMicro) (1.3.1)
Requirement already satisfied: python-dateutil>=2.7 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from matplotlib->BMicro) (2.8.1)
Requirement already satisfied: pyparsing>=2.2.1 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from matplotlib->BMicro) (2.4.7)
Requirement already satisfied: fonttools>=4.22.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from matplotlib->BMicro) (4.29.1)
Requirement already satisfied: pillow>=6.2.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from matplotlib->BMicro) (8.1.0)
Requirement already satisfied: six in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from cycler>=0.10->matplotlib->BMicro) (1.15.0)
Requirement already satisfied: networkx>=2.2 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from scikit-image->BMicro) (2.5)
Requirement already satisfied: tifffile>=2019.7.26 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from scikit-image->BMicro) (2021.1.14)
Requirement already satisfied: PyWavelets>=1.1.1 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from scikit-image->BMicro) (1.1.1)
Requirement already satisfied: imageio>=2.4.1 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from scikit-image->BMicro) (2.9.0)
Requirement already satisfied: decorator>=4.3.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from networkx>=2.2->scikit-image->BMicro) (4.4.2)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pyqt5->BMicro) (12.8.1)
Requirement already satisfied: iniconfig in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pytest->bmlab->BMicro) (1.1.1)
Requirement already satisfied: tomli>=1.0.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pytest->bmlab->BMicro) (2.0.1)
Requirement already satisfied: pluggy<2.0,>=0.12 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pytest->bmlab->BMicro) (1.0.0)
Requirement already satisfied: py>=1.8.2 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pytest->bmlab->BMicro) (1.11.0)
Requirement already satisfied: attrs>=19.2.0 in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from pytest->bmlab->BMicro) (21.2.0)
WARNING: You are using pip version 21.1.1; however, version 22.0.3 is available.
You should consider upgrading via the '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3 -m pip install --upgrade pip' command.
01-guc-grp-0050:Desktop jkolb$ python3 -m bmicro
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/runpy.py", line 87, in _run_code
[... cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bmicro/gui/calibration/__init__.py", line 2, in <module>
    from .calibration_view import CalibrationView
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bmicro/gui/calibration/calibration_view.py", line 13, in <module>
    from bmlab.controllers.calibration_controller import CalibrationController
ModuleNotFoundError: No module named 'bmlab.controllers.calibration_controller'; 'bmlab.controllers' is not a package

raimund-schluessler, 2022-02-17:
@juko1 As stated, please use the DMG file from the release and not pip for now.

raimund-schluessler, 2022-02-17:
@paulmueller Do you have any idea what could cause this problem? It runs fine like this on Windows.

juko1, 2022-02-17:
The dmg file opens repeatedly...

raimund-schluessler, 2022-02-17:
> The dmg file opens repeatedly...
> 
> 

What do you mean by that?

juko1, 2022-02-17:
it opens normally but than opens another window and again and again until I shut down the computer...

paulmueller, 2022-02-17:
It could be because `controllers` is not imported in `bmlab/__init__.py`.

paulmueller, 2022-02-17:
When I look at the bmlab source, I think it should be 

```
from bmlab.controllers import CalibrationController
```

instead of

```
 from bmlab.controllers.calibration_controller import CalibrationController
```

Maybe an old version of bmlab is installed?

@juko1 Could you please try `pip install --upgrade bmlab bmicro`?

@raimund-schluessler If you have breaking changes you should bump the minium version requirement in setup.py. (e.g. `install_requires="bmlab>=0.12.0"` in https://github.com/BrillouinMicroscopy/BMicro/blob/main/setup.py#L39.

raimund-schluessler, 2022-02-17:
> Maybe an old version of bmlab is installed?

Yes, indeed, this must be it. Looking at the log, I now also see it installs bmlab 0.0.11, which does not work with BMicro@latest anymore.

> @raimund-schluessler If you have breaking changes you should bump the minium version requirement in setup.py. (e.g. `install_requires="bmlab>=0.12.0"` in https://github.com/BrillouinMicroscopy/BMicro/blob/main/setup.py#L39.

Thanks for the hint, I will do that. I just assumed it automatically installs the latest version available.

> ```
>  from bmlab.controllers.calibration_controller import CalibrationController
> ```

I found that weird already, but I just assumed Python does some weird CamelCase -> snake_case conversion.

raimund-schluessler, 2022-02-17:
Or actually, it's the other way around. pip installs BMicro 0.1.3 which is to old to work with bmlab 0.0.11. Either way
`pip install --upgrade bmlab bmicro` should help fix it for now.

juko1, 2022-02-17:
Nice, now, it works! I had to enter 'python3 -m pip install --upgrade bmlab bmicro'

closed 2022-02-17 by raimund-schluessler

juko1, 2022-03-17:
same problem with version 0.2.0, the dmg file opens repeatedly. This time 'python3 -m pip install --upgrade bmlab bmicro' didn't helped

raimund-schluessler, 2022-03-17:
Yes, the DMG file is still broken. I try to fix it in #136.
What do you mean with `python3 -m pip install --upgrade bmlab bmicro` didn't help? The DMG file and installing via pip is kind of unrelated.

PR #122 (2022-02-17): Request minimum required bmlab version
Closes #112.