BrillouinMicroscopy/BMicro #113: error even the file can be loaded

juko1, 2022-02-15:
[image]
[image]

raimund-schluessler, 2022-02-15:
In my case this happens sometimes if the file I try to open is located on a network drive (so it might take longer to open). The evaluation works fine nevertheless.
It never happens with a file locally on the PC.

raimund-schluessler, 2022-02-17:
I don't see this behaviour with 0.1.5. anymore. @juko1 Could you please test if this still happens for you with the latest release?

raimund-schluessler, 2022-03-16:
I can't reproduce this anymore. Please reopen if this still occurs for you.

closed 2022-03-16 by raimund-schluessler