BrillouinMicroscopy/BMicro #123: Installing BMicro via pip on MacOSX Monterey fails

raimund-schluessler, 2022-02-18:
Installing BMicro on MacOSX Monterey results in:

```bash
python3 -m pip install BMicro
 Defaulting to user installation because normal site-packages is not writeable
Collecting BMicro
  Using cached bmicro-0.1.5-py3-none-any.whl (43 kB)
Collecting matplotlib
  Using cached matplotlib-3.5.1-cp38-cp38-macosx_11_0_arm64.whl (7.2 MB)
Collecting scipy>=0.14.0
  Downloading scipy-1.8.0-cp38-cp38-macosx_12_0_arm64.whl (28.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 28.5/28.5 MB 24.4 MB/s eta 0:00:00
Collecting pytest
  Downloading pytest-7.0.1-py3-none-any.whl (296 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 297.0/297.0 KB 8.2 MB/s eta 0:00:00
Collecting h5py>=2.10.0
  Using cached h5py-3.6.0.tar.gz (384 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pytest-qt
  Downloading pytest_qt-4.0.2-py2.py3-none-any.whl (33 kB)
Collecting pyqtgraph
  Downloading pyqtgraph-0.12.3-py3-none-any.whl (992 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 992.4/992.4 KB 19.8 MB/s eta 0:00:00
Collecting pytest-mock
  Downloading pytest_mock-3.7.0-py3-none-any.whl (12 kB)
Collecting scikit-image
  Downloading scikit_image-0.19.2-cp38-cp38-macosx_12_0_arm64.whl (12.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 12.3/12.3 MB 25.9 MB/s eta 0:00:00
Collecting bmlab
  Downloading bmlab-0.0.12-py3-none-any.whl (42 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.0/43.0 KB 1.5 MB/s eta 0:00:00
Collecting pyqt5
  Using cached PyQt5-5.15.6.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Traceback (most recent call last):
        File "/Users/vaibhavmahajan/Library/Python/3.8/lib/python/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 156, in prepare_metadata_for_build_wheel
          hook = backend.prepare_metadata_for_build_wheel
      AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'

      
      During handling of the above exception, another exception occurred:

      
      Traceback (most recent call last):
        File "/Users/vaibhavmahajan/Library/Python/3.8/lib/python/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/Users/vaibhavmahajan/Library/Python/3.8/lib/python/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/Users/vaibhavmahajan/Library/Python/3.8/lib/python/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 160, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/private/var/folders/23/w4k3qp2n22j32fzwqxcbqv0h0000gq/T/pip-build-env-_gzxqaoc/overlay/lib/python3.8/site-packages/sipbuild/api.py", line 51, in build_wheel
          project = AbstractProject.bootstrap('pep517')
        File "/private/var/folders/23/w4k3qp2n22j32fzwqxcbqv0h0000gq/T/pip-build-env-_gzxqaoc/overlay/lib/python3.8/site-packages/sipbuild/abstract_project.py", line 83, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/23/w4k3qp2n22j32fzwqxcbqv0h0000gq/T/pip-build-env-_gzxqaoc/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 594, in setup
          self.apply_user_defaults(tool)
        File "project.py", line 63, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/23/w4k3qp2n22j32fzwqxcbqv0h0000gq/T/pip-build-env-_gzxqaoc/overlay/lib/python3.8/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/23/w4k3qp2n22j32fzwqxcbqv0h0000gq/T/pip-build-env-_gzxqaoc/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 241, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/23/w4k3qp2n22j32fzwqxcbqv0h0000gq/T/pip-build-env-_gzxqaoc/overlay/lib/python3.8/site-packages/pyqtbuild/builder.py", line 67, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]

  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

paulmueller, 2022-02-18:
`Building h5py requires pkg-config unless the HDF5 path is explicitly specified` - I think that means that maybe on that intel silicon machine the HDF5 path is specified for some reason? Sounds like an h5py packaging issue.

raimund-schluessler, 2022-03-24:
Since the problem is with h5py and we have a workaround (installing BMicro via the DMG packages provided), I am going to close this.

closed 2022-03-24 by raimund-schluessler