BrillouinMicroscopy/BMicro #139: BMicro shuts down after clicking evaluate

juko1, 2022-03-18:
/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 11 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

raimund-schluessler, 2022-03-18:
Thanks. Is this the complete warning/error message? I have never seen it honestly.
But as this again is an issue with multiprocessing, see #140, I tend to disable multiprocessing alltogether until we can handle it properly. This will slowdown the evaluation a bit (I think it was factor 2), but at least makes the program usable again (I hope).
@paulmueller Feedback is very welcome inn case you have an idea or more experience with multiprocessing.

juko1, 2022-03-18:
I've got a problem report for Python as well. I'll send you that via email, because it's too long for this comment window.

paulmueller, 2022-03-18:
Sorry, that's too technical for me. Maybe you can find something on stackoverflow? Have tried upgrading PyInstaller?

raimund-schluessler, 2022-03-18:
> Sorry, that's too technical for me. Maybe you can find something on stackoverflow? Have tried upgrading PyInstaller?

PyInstaller is the latest version already. These multiprocessing errors seem a bit obscure and difficult to solve. I will check further, but disable multiprocessing for now.

raimund-schluessler, 2022-03-21:
Closed by https://github.com/BrillouinMicroscopy/bmlab/pull/81

closed 2022-03-21 by raimund-schluessler