BrillouinMicroscopy/BMicro #141: BMicro shuts down when it cannot find the peaks for all calibrations

juko1, 2022-03-18:
Select calibration peaks in Extraction mode: When clicking 'find peaks for all calibrations' bmicro shuts down immediately. This happened only with some files. Selecting the calibration peaks separately works.
smb://syn01.frontend.storage.int.mpl.mpg.de/guck_division2/Data/FOB/Julia/20220317_Fmoda elavl-GFP 1dpl/RawData/Brillouin-10.h5.zip

sorry, I cannot upload the zip file. It says 'is not included in the list' - what does that mean?

raimund-schluessler, 2022-03-18:
Thanks for the report. But I don't have access to the internal smb server. Could you please upload it to a public sharing software (owncloud, nextcloud or whatever mpl uses). You can also send me the link via rocketchat if you like.

raimund-schluessler, 2022-03-18:
The problem arises from the first three points found laying on the same line. Then bmlab cannot fit a circle and crashes.

closed 2022-03-18 by raimund-schluessler