BrillouinMicroscopy/BMicro #150: Cannot load evaluated repetition (via BMicro dmg) into impose

juko1, 2022-03-23:
I evaluated the 2nd repetition of one measurement with the BMicro app. But I cannot load this file into impose (see error report below). There was also no .mat file created, is there a connection?
Link for the raw data will come per Mail.

Unhandled exception in Impose version 0.2.0:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/gui/colocalize.py", line 174, in on_add_data
    self.add_paths(paths)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/gui/colocalize.py", line 105, in add_paths
    self.session_scheme.append(pp)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/session.py", line 155, in append
    data_source = DataSource(path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/data.py", line 43, in __init__
    self._initialize_from_path(path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/data.py", line 89, in _initialize_from_path
    data_channels, meta_orig = load(path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/formats/__init__.py", line 45, in load
    channels, meta = suffix_dict[path.suffix](path)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/formats/fmt_bm_bmlab.py", line 66, in load_h5
    axes = np.string_(h5[k0].attrs["axisOrder"]).decode("utf-8")
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/h5py/_hl/attrs.py", line 56, in __getitem__
    attr = h5a.open(self._id, self._e(name))
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5a.pyx", line 80, in h5py.h5a.open
KeyError: "Can't open attribute (can't locate attribute: 'axisOrder')"

juko1, 2022-03-25:
Unhandled exception in Impose version 0.2.2:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/gui/colocalize.py", line 194, in on_dataset_selected
    self.update_ui_from_scheme()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/gui/colocalize.py", line 336, in update_ui_from_scheme
    self.rois.set_structure_composite(self.current_structure_composite,
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/gui/colocalize_pgrois.py", line 170, in set_structure_composite
    tr = vis.get_roi_transform(roi)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/impose/gui/visualize.py", line 150, in get_roi_transform
    data = np.array(self.imageView.image)[:, :, 0]
IndexError: too many indices for array: array is 0-dimensional, but 3 were indexed

paulmueller, 2022-03-25:
@raimund-schluessler I'm confused, so I am reopening this.

raimund-schluessler, 2022-03-25:
> @raimund-schluessler I'm confused, so I am reopening this.

It's another error, unrelated to the one intially reported here. It is due to how Impose handles NaN values. Apparently my fix in https://github.com/GuckLab/impose/pull/27 was incomplete. Follow-up is in https://github.com/GuckLab/impose/pull/34.

@juko1 Please report issues with Impose in the correct repository https://github.com/GuckLab/impose

raimund-schluessler, 2022-04-11:
I moved the issue over to https://github.com/GuckLab/impose/issues/36 as it's a problem with Impose.

closed 2022-04-11 by raimund-schluessler